OpenTabletDriver/OpenTabletDriver #3832: Can't install Plugins

skyisthela, 2025-03-06:
### Steps to Reproduce

Install OpenTabletDriver from software manager
Go to Plugin Manager in the program 
Install Plugin
Nothing happens

### Expected Behavior

Plug in gets installed

### Observed Behavior

It doesn't get installed

### Tablet Device

CTL-472

### Diagnostic Information

[ds.json](https://github.com/user-attachments/files/19105716/ds.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-03-06:
Can't reproduce? Video? Anything more concrete?

skyisthela, 2025-03-06:
> Can't reproduce? Video? Anything more concrete?

well, a whole new issue popped up. I reinstalled OTD but instead of the plugins not being able to be installed, if I try to change the area it doesn't work at all. But if I restart OTD the issue comes back again. I tried basic trouble shooting with like the wiki suggested but nothing works. If I restart my pc everything including the area gets reset and the issue still persists.

https://files.catbox.moe/4eubs0.mp4

jamesbt365, 2025-03-06:
For whatever reason, the directory OTD is trying to write to is not writeable.

Also, the flatpak package is not an official package so we don't provide support for it.

closed 2025-03-06 by jamesbt365

skyisthela, 2025-03-06:
> For whatever reason, the directory OTD is trying to write to is not writeable.
> 
> Also, the flatpak package is not an official package so we don't provide support for it.

the reason why i use this instead of the one directly installed from the site is because if i use that it keeps telling me connection timed out daemon not detected blah blah blah

jamesbt365, 2025-03-06:
> keeps telling me connection timed out daemon not detected

[image]

skyisthela, 2025-03-06:
> > keeps telling me connection timed out daemon not detected
> 
> [image]

https://files.catbox.moe/f4i9jr.mp4 lmao

jamesbt365, 2025-03-06:
Stop running multiple drivers at once.

skyisthela, 2025-03-06:
> Stop running multiple drivers at once.

i'm not, the moment i boot up the computer this happens. the moment i click on the tablet it creates a duplicate and they start blinking.

jamesbt365, 2025-03-06:
https://opentabletdriver.net/Wiki/Documentation/RequiredPermissions#package-manager

skyisthela, 2025-03-06:
> https://opentabletdriver.net/Wiki/Documentation/RequiredPermissions#package-manager

thank you so much this fixed it