OpenTabletDriver/OpenTabletDriver #3838: Add support for Huion Kamva 13 2022

shawny43, 2025-03-12:
### Manufacturer

Huion

### Product Name

GT-133

### Device Width

388

### Device Height

219

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Here the manual to know more.

[User Manual_Kamvas Pro 13.pdf](https://github.com/user-attachments/files/19202298/User.Manual_Kamvas.Pro.13.pdf)

### Diagnostic Information

[diag.json](https://github.com/user-attachments/files/19202328/diag.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

shawny43, 2025-10-07:
> Please unload the stock tablet driver entirely and upload a new diagnostics file.
> 
> Additionally, we'll need a full string dump of the tablet, since Huion tablets differentiate themselves on strings. You can get this from the driver GUI via Tablets -> Device String Reader - fill in VendorID `9580` and ProductID `109`.

It's return to invalid index, no journal entries and also i have all the logs of execution in otd-daemon from terminal. 

But fortunately the tablet have been recognized. but with limited options since the touchbar slider are not available. 

<details>
<summary> OTD-Daemon logs execution </summary>
[Plugin:Debug]	Loading plugin 'Windows Ink'
[Plugin:Debug]	Loading plugin 'Tilt Calibration'
[Plugin:Debug]	Loading plugin 'Tablet Calibration'
[Plugin:Debug]	Loading plugin 'MouseUtils'
[Plugin:Debug]	Loading plugin 'BetterCalibrator'
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Searching for tablet 'Acepen AP1060'
[Detect:Debug]	Searching for tablet 'Acepen AP906'
[Detect:Debug]	Searching for tablet 'Adesso Cybertablet K8'
[Detect:Debug]	Searching for tablet 'Artisul A1201'
[Detect:Debug]	Searching for tablet 'Artisul AP604'
[Detect:Debug]	Searching for tablet 'Artisul D16 Pro'
[Detect:Debug]	Searching for tablet 'Artisul M0610 Pro'
[Detect:Debug]	Searching for tablet 'FlooGoo FMA100'
[Detect:Debug]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Debug]	Searching for tablet 'Gaomon GM116HD'
[Detect:Debug]	Searching for tablet 'Gaomon GM156HD'
[Detect:Debug]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M106K'
[Detect:Debug]	Searching for tablet 'Gaomon M10K Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M10K'
[Detect:Debug]	Searching for tablet 'Gaomon M1220'
[Detect:Debug]	Searching for tablet 'Gaomon M1230'
[Detect:Debug]	Searching for tablet 'Gaomon M5'
[Detect:Debug]	Searching for tablet 'Gaomon M6'
[Detect:Debug]	Searching for tablet 'Gaomon M8 (Variant 2)'
[Detect:Debug]	Searching for tablet 'Gaomon M8'
[Detect:Debug]	Searching for tablet 'Gaomon PD1161'
[Detect:Debug]	Searching for tablet 'Gaomon PD156 Pro'
[Detect:Debug]	Searching for tablet 'Gaomon PD1560'
[Detect:Debug]	Searching for tablet 'Gaomon PD1561'
[Detect:Debug]	Searching for tablet 'Gaomon PD2200'
[Detect:Debug]	Searching for tablet 'Gaomon S56K'
[Detect:Debug]	Searching for tablet 'Gaomon S620'
[Detect:Debug]	Searching for tablet 'Gaomon S630'
[Detect:Debug]	Searching for tablet 'Gaomon S830'
[Detect:Debug]	Searching for tablet 'Genius G-Pen 560'
[Detect:Debug]	Searching for tablet 'Genius i405x'
[Detect:Debug]	Searching for tablet 'Genius i608x'
[Detect:Debug]	Searching for tablet 'Huion 1060 Plus'
[Detect:Debug]	Searching for tablet 'Huion 420'
[Detect:Debug]	Searching for tablet 'Huion G10T'
[Detect:Debug]	Searching for tablet 'Huion G930L'
[Detect:Debug]	Searching for tablet 'Huion GC610'
[Detect:Debug]	Searching for tablet 'Huion GT-156HD V2'
[Detect:Debug]	Searching for tablet 'Huion GT-191 V2'
[Detect:Debug]	Searching for tablet 'Huion GT-220 V2'
[Detect:Debug]	Searching for tablet 'Huion GT-221 Pro'
[Detect:Debug]	Searching for tablet 'Huion GT-221'
[Detect:Debug]	Searching for tablet 'Huion H1060P'
[Detect:Debug]	Searching for tablet 'Huion H1061P'
[Detect:Debug]	Searching for tablet 'Huion H1161'
[Detect:Debug]	Searching for tablet 'Huion H320M'
[Detect:Debug]	Searching for tablet 'Huion H420'
[Detect:Debug]	Searching for tablet 'Huion H420X'
[Detect:Debug]	Searching for tablet 'Huion H430P'
[Detect:Debug]	Searching for tablet 'Huion H580X'
[Detect:Debug]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Debug]	Searching for tablet 'Huion H610 Pro V3'
[Detect:Debug]	Searching for tablet 'Huion H610 Pro'
[... 248 lines cut ...]
[Detect:Debug]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430S V2'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G540'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640 (Variant 2)'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S'
[Evdev:Debug]	Successfully initialized virtual pressure sensitive tablet. (code NONE)
[Huion Kamvas Pro 13:Info]	Output mode: Artist Mode
[Huion Kamvas Pro 13:Info]	Display area: [6160x1440@<3080  720>:0°],
[Huion Kamvas Pro 13:Info]	Tablet area: [293,76x165,24@<146,88  82,62>:0°],
[Huion Kamvas Pro 13:Info]	Clipping: Enabled
[Huion Kamvas Pro 13:Info]	Ignoring reports outside area: Disabled
[Huion Kamvas Pro 13:Info]	Pressure: Enabled
[Huion Kamvas Pro 13:Info]	Tilt: Enabled
[Huion Kamvas Pro 13:Info]	Tip Binding: [Adaptive Binding: Tip]@1%
[Huion Kamvas Pro 13:Info]	Eraser Binding: [Adaptive Binding: Eraser]@1%
[Huion Kamvas Pro 13:Info]	Pen Bindings: Adaptive Binding: Button 1, Adaptive Binding: Button 2
[Settings:Info]	Driver is enabled.
</details>

shawny43, 2025-12-15:
Hello, it's a while we didn't talk about the issue but has i saw, my tablet have been added to the list of supported but still not detected natively on the OTD. Can we explain why?

gonX, 2026-10-04:
You need to ensure the vendor driver is unloaded. Since you're using the Flatpak, you need to follow its description closely.

Also, Huion's naming scheme is quite confusing. It is not clear if your tablet is supported yet - how are you establishing that it should be supported now?

Puhama, 2026-10-22:
@gonX 
I am not OP but I also own the Huion Camvas Pro 3. The tablet is detected without problems but it is not working with Wayland (x11 works).
I am using CachyOS.