OpenTabletDriver/OpenTabletDriver #3849: Relative Mode Pressure Sensitivity Support

aj-ru, 2025-03-26:
### Description

It seems like the relative mode on OTD does not have pressure sensitivity support, at least on the One by Wacom. Wacom's official driver software as well doesn't support pressure for "Pen Mode".

I want some help understanding why this is the case. I use my small One by Wacom for notes. On Absolute/Artist Mode, since the tablet is small, I have to zoom in to the level of a sentence in order to take notes on my small tablet. [This video](https://www.youtube.com/watch?v=35Xoj041ePc) I found on YouTube does so.

However, by using relative mode, not only can I comfortably view the whole page while taking notes, I can always place my hand at the center of the tablet, avoiding the discomfort of writing at the edges of the tablet. 

To move the cursor forward, I just have to move the tip forward, do a motion with my fingers to take it out of range of the tip detector, and bring it down back to the original position, only this time with the cursor being advanced!

I discovered this trick within a day of getting the Wacom, and I assumed that this is how people took notes on the Wacom. However, this does not seem to be the case, and I'm utterly baffled, perplexed, and truly shocked.

So, coming to my question: it seems that relative mode is used as a . . . graphic tablet replacement for a **mouse**, not for notes. And in that use case, sure, pressure sensitivity might not make a lot of sense. But . . . pressure sensitivity when taking *notes* makes the handwriting look more "real" and "personal". So please considering adding pressure support for the relative mode. Please.

GNOME on Linux supports this. However, since it doesn't allow adjusting the speed of the stylus, I have to zoom in a lot more, which is why I use OTD, which offers movement sensitivity support.

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-12-05:
This is going to be a bit complicated to properly fix. Surprisingly, removing `INPUT_PROP_DIRECT` from Artist Mode's code still makes it initialize as an absolutely positioned tablet. Looking at systemd's udev code (`src/udev/udev-builtin-input_id.c`), it looks something like this:

```c
        if (has_abs_coordinates) {
                if (has_stylus || has_pen)
                        is_tablet = true;
        }
```

So instead we would need to _duplicate_ artist mode, change it to use `REL_X`/`REL_Y` and pray that it still works, or use an udev file to force the device to show up as a mouse instead of a tablet. All of this would need further research into the specifics of each supported OS. MacOS for example does already have pressure sensitivity on its relative mode output mode.

So the preferred fix is that we do this abstraction inside of OpenTabletDriver instead. This is likely not happening for the next release cycle.

swindlesmccoop, 2026-10-28:
I made a plugin that adds this functionality on Linux:

https://github.com/swindlesmccoop/RelativeArtistMode

Some features are only tested on KDE but it should work on GNOME as well. Full transparency: it was heavily Claude-assisted; if you find that to be a problem, feel free to not use it. I'm not sure how the official plugin repo feels about AI-assisted code submissions, so you'll have to manually install it. I've already built a release, so you just have to download the zip and install it in OTD. Then once you switch mode, it should just work. I've been using it for a few days now with no issues.

KiraGreifeneder, 2026-10-28:
@swindlesmccoop I’ll be trying it out, that sounds great. Thanks for being transparent about the AI usage too. The contribution guidelines say that AI code is not strictly banned, but it must be disclosed. 

Would love to see this built directly into OTD eventually.