OpenTabletDriver/OpenTabletDriver #3853: cant use windows ink with Wacom CTL-6100WL

Slaus890, 2025-03-30:
### Steps to Reproduce

I have installed VMulti driver and windows ink driver. But when i try it in krita it just straight up doesn`t work, it worked with original wacom drivers but i switched to this one since this one is a lot more customizable

### Expected Behavior

to pen pressure work in krita/

### Observed Behavior

pen pressure doesn`t work in krita

### Tablet Device

CTL-6100WL

### Diagnostic Information

[thing.json](https://github.com/user-attachments/files/19525470/thing.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Slaus890, 2025-03-30:
i`m sorry if i wrote this in the wrong article.

jamesbt365, 2025-03-30:
https://opentabletdriver.net/Wiki/FAQ/Windows#win-ink

closed 2025-03-30 by jamesbt365