OpenTabletDriver/OpenTabletDriver #3866: Add support for JSK-DP22/DP23

timelessnesses, 2025-04-11:
### Manufacturer

JSKPAD

### Product Name

DP22/DP23

### Device Width

200mm

### Device Height

141mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

the [tablet](https://www.dgjskpad.com/digital-tablet-p62950218.html) also has it's own driver which can be downloaded [here](https://www.dgjskpad.com/download.html)  
the tablet also has 6 keybind buttons and 2 buttons on the pen

### Diagnostic Information

[diagnostics.json](https://github.com/user-attachments/files/19714532/w.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-04-11:
tablets > device string reader
10473
7
leave blank
save that file, send it here.

timelessnesses, 2025-04-12:
> tablets > device string reader 10473 7 leave blank save that file, send it here.

[image]  
like this? and click on dump all?  

[wh.txt](https://github.com/user-attachments/files/19715346/wawa.txt)