OpenTabletDriver/OpenTabletDriver #3869: Create support for GT1060p parrot 10x6 inch graphics tablet

DoomedButton327, 2025-04-16:
Manufacturer
Parrot

Product Name
GT1060P

Device Width
360mm

Device Height
240mm

Active area
10x6.25 inches

Pressure Levels
8192

Lines per Inch (LPI)
5080

Additional Information
the tablet also has it's own driver which can be downloaded [here](https://www.parrot.co.za/product/sku/GT1060P?srsltid=AfmBOoqNfeJ5aygEKmtS_EmJdzDYrXfLGy-sE8Hh8rYMs6dFPABsa8xV&graphics-tablet-wired-10-x-6-inch)
the tablet also has 12 keybind buttons+10 media buttons(like audio,calculator buttons) and 2 buttons on the pen

Diagnostic Information
Don't have any diagnostics info yet

Added comment
Hope this is better than the last one, the other one didn't give me 0 templates but said it had given me one

jamesbt365, 2025-04-16:
> Don't have any diagnostics info yet

This is a requirement and we cannot proceed without it. I will reopen when you attach diagnostics.

> Hope this is better than the last one, the other one didn't give me 0 templates but said it had given me one

You are prompted when you are going to open the issue.

DoomedButton327, 2025-04-17:
I have attached the diagnostics, sorry for late reply, have being busy

jamesbt365, 2025-04-17:
This is in the same boat as the 10moons tablet we recently removed support for, but I'll reopen anyway despite the chances of this ever being supported being extremely slim.