OpenTabletDriver/OpenTabletDriver #3870: Experiencing a "vertical crease" in my pen-to-cursor translation (Wacom PTK-440, Absolute Mode, Locked Aspect Ratio, 12-16mm width, M1 MacOS Sonoma 14.4.1)

rasputozen, 2025-04-17:
### Steps to Reproduce

1. Settings: Absolute Mode, Wacom PTK-440, Locked Aspect Ratio, Width 12mm (also tried 16mm)

### Expected Behavior

I expected a distortion-less mapping from pen to cursor.

### Observed Behavior

Move cursor around and notice a vertical "crease" or "fold" in the mapping from pen to cursor. A perfectly horizontal movement of the pen causes the cursor to shift up or down vertically as the pen crosses this invisible crease/fold.
I tried moving where the 12mm section was mapped onto the tablet (to test if the defect was the tablet itself) and it had no noticeable effect on this distortion, except maybe causing the crease to be offset to a somewhat different place. Changing from 12mm to 16mm width caused the crease to change position though both were somewhere around the middle.

### Tablet Device

Wacom PTK-440

### Diagnostic Information

[otd-diagnostics.json](https://github.com/user-attachments/files/19787112/otd-diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-04-17:
tablets > tablet debugger, see if the problem happens on the visualizer.

jamesbt365, 2025-04-17:
Could you record it?

rasputozen, 2025-04-17:
> Could you record it?

https://www.youtube.com/watch?v=ort3PHCCjcQ

rasputozen, 2025-04-20:
Has this been identified as a bug?