OpenTabletDriver/OpenTabletDriver #3871: XP-Pen Deco 01 V2 (Variant 2) not correctly detected resulting in wrong active area

TecnoCreeper, 2025-04-17:
### Steps to Reproduce

- Plug tablet XP-Pen 01 V2 (Variant 2)
- Start OpenTabletDriver
- Set tablet to full area

### Expected Behavior

The active area should be the full size of the tablet.

### Observed Behavior

The active area is not the full size of the tablet. It is roughly 2/3 the width and 1/2 the height.

### Tablet Device

XP-Pen Deco 01 V2 (Variant 2)

### Diagnostic Information

[diagnostics.json](https://github.com/user-attachments/files/19796279/diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

TecnoCreeper, 2025-04-17:
#3678 fixes the problem for me.

tujh2, 2025-12-05:
fixed by https://github.com/OpenTabletDriver/OpenTabletDriver/pull/4293

TecnoCreeper, 2025-12-05:
(Running version 0.6.6.2) Using the configuration overrides from #4293 also fixes the issue.
Will close the issue when the fix is in a release and I verify it works correctly.

gonX, 2025-12-30:
Closing as fixed by #4293 - we close issues based on whether a fixing pull request is merged. You can try a GitHub Actions build if you want to test a pre-release version of it working.