OpenTabletDriver/OpenTabletDriver #3873: Express keys not working on Wacom CTH-470

fidoboy, 2025-04-19:
### Steps to Reproduce

There is a warning message on logs: "[Device:Warning] Failed to set device feature: 02-02" and the four express keys on this tablet are not working.

### Expected Behavior

I expected that this works out of the box

### Observed Behavior

The four buttons on the tablet does nothing

### Tablet Device

Wacom CTH-470

### Diagnostic Information

[export.json](https://github.com/user-attachments/files/19822248/export.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-04-19:
You have another driver installed.

fidoboy, 2025-04-19:
It's 6.5.1 as says in the OpenTablet window

fidoboy, 2025-04-19:
The export file also says: OpenTabletDriver v0.6.5.1 so what "other driver" are you talking about?

jamesbt365, 2025-04-19:
> It's 6.5.1 as says in the OpenTablet window

No, as in you have multiple, OTD can't initialise and use the device because it's being used by another. You cannot have multiple tablet drivers installed at once.

fidoboy, 2025-04-19:
So should I uninstall the default Wacom driver installed by Windows?

jamesbt365, 2025-04-19:
> So should I uninstall the default Wacom driver installed by Windows?

Yes, see our FAQ for more details.

fidoboy, 2025-04-22:
> > So should I uninstall the default Wacom driver installed by Windows?
> 
> Yes, see our FAQ for more details.

This issue is not solved.

I've followed the instructions. I've executed the tabletdriver uninstaller and also Driver Store Explorer to ensure that there are no drivers from Wacom or whatever in my system. Then I restarted the computer and plugged my tablet. I opened the OpenTabletDriver app and my tablet was detected fine, it displays the model and manufacturer. But express keybindings are not working. 

To test it I've binded the keys A, B, C and D to each of the four buttons on the side of the tablet. Then I opened notepad to test it by pressing each tablet button. No work. There is no output. Can't see also anything on the OpenTabletDriver log tab. 

As a suggestion, it could be very useful for future releases of this project, to have any tab into the OpenTabletDriver app interface where it can display visually any tablet output. Something simmilar to the gamepads configuration utilities where you can see buttons pressed and stick movements. Take this as my two cents to this amazing project.

But for now, I can't understand why express key bindings are not working. Now the "Failed to set device feature: 02-02" message has disappeared but however the keys are not working.

fidoboy, 2025-04-22:
Also I noticed that here it's instructed to install the VMulti driver for Windows Ink to work. What should I do about it? Do I need to install it or not? This is because you said that it's needed that NO DRIVER was installed on the system. So I'm not sure abou what to do about this. For the moment, I've not installed it because first I need to have express keys working.

gonX, 2025-04-22:
Change the diagnostics in the original post to a fresh one after you've ensured that the Wacom driver is uninstalled, then we can reopen the issue.

fidoboy, 2025-04-22:
> Change the diagnostics in the original post to a fresh one after you've ensured that the Wacom driver is uninstalled, then we can reopen the issue.

Ok, thanks. I'll do it inmediately.

jamesbt365, 2025-04-22:
> As a suggestion, it could be very useful for future releases of this project, to have any tab into the OpenTabletDriver app interface where it can display visually any tablet output.

Tablets > tablet debugger

> Now the "Failed to set device feature: 02-02" message has disappeared but however the keys are not working.

Great! Get a fresh diagnostic so we can check it out.

> it's instructed to install the VMulti driver for Windows Ink to work. What should I do about it?

Install it

> This is because you said that it's needed that NO DRIVER was installed on the system. 

Vmulti is not a tablet driver.

fidoboy, 2025-04-22:
I've updated the first post with a new report after I've ran tablet driver uninstaller

jamesbt365, 2025-04-22:
Tablets > tablet debugger, press the tablet buttons what do you see?

fidoboy, 2025-04-22:
> Tablets > tablet debugger
>
>
> Great! Get a fresh diagnostic so we can check it out.
>
>
> Install it
>
>
> Vmulti is not a tablet driver.

Thanks @jamesbt365 for your kindly indications. I've tried the tablet debuger interface. It's not very user friendly, a bit technical, but it's fine. I was talking about something more user friendly with a simple picture like the gamepads in some emulator applications has. But it doesn't mind. I suppose it's fine as it is.

Using that tablet debugger I've noticed that when I press my tablet keys it's being reflected in the form of a "true" or something like that (I barely can see it because it's displayed for less than a second and then disappears)

jamesbt365, 2025-04-22:
Well then, the auxiliary keys are working properly and if the buttons are not actually being pressed then this is due to a misconfiguration of settings, either from a plugin or user error.

File > reset to defaults, set your bindings, save.

If they work then, but stop working after you enable/change other stuff, theres your answer.

fidoboy, 2025-04-22:
I assume that installing VMulti is fine then. So I'll install it as I did it before to have Windows Ink working. I'll try now to reset all key bindings again to check if it could work.

fidoboy, 2025-04-22:
For the record, I want to notice here that when I opened the tablet debugger the log displayed this: "[TabletDebugger:Warning]	Touch undefined in tablet configuration - unable to draw touch points" Can't understand why.

jamesbt365, 2025-04-22:
> For the record, I want to notice here that when I opened the tablet debugger the log displayed this: "[TabletDebugger:Warning] Touch undefined in tablet configuration - unable to draw touch points" Can't understand why.

Because touch is undefined in the tablet configuration because OTD doesn't support touch output.

fidoboy, 2025-04-22:
I've cleared all resetting the app to defaults and now express keys seems to be working fine. Thanks a lot for your patience. Now I'll try to install the VMulti driver and let's see what happens.

closed 2025-04-22 by jamesbt365

fidoboy, 2025-04-22:
I also hope that this steps I followed here could be helpful for other users wich reach this thread.

fidoboy, 2025-04-22:
This is what it's being displayed into the log when I open the app now:

```
[Detect:Info]	Found tablet 'Wacom CTH-470'
[Wacom CTH-470:Info]	Output mode: Windows Ink Absolute Mode
[Wacom CTH-470:Info]	Display area: [1920x1200@<2640. 600>:0°],
[Wacom CTH-470:Info]	Tablet area: [147,2x92@<73,6. 46>:0°],
[Wacom CTH-470:Info]	Clipping: Enabled
[Wacom CTH-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTH-470:Info]	Tip Binding: [VoiDPlugins.OutputMode.WindowsInkButtonHandler]@1%
[Wacom CTH-470:Info]	Eraser Binding: [VoiDPlugins.OutputMode.WindowsInkButtonHandler]@1%
[Wacom CTH-470:Info]	Pen Bindings: VoiDPlugins.OutputMode.WindowsInkButtonHandler, VoiDPlugins.OutputMode.WindowsInkButtonHandler
[Wacom CTH-470:Info]	Express Key Bindings: Key Binding: A, Key Binding: B, Key Binding: C, Key Binding: D
[Wacom CTH-470:Info]	Filters: BezierInterpolator.BezierInterp, BetterCalibrator.BetterCalibrator
[Settings:Info]	Driver is enabled.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
```

The four express keys are not working now when it's in Windows Ink mode. But however I can see that it works in tablet debugger interface because it displays a "true" for a little instant.

jamesbt365, 2025-04-22:
Need another diagnostic.

fidoboy, 2025-04-22:
Here it's 

[report.json](https://github.com/user-attachments/files/19849491/report.json)

jamesbt365, 2025-04-22:
I notice you have filters enabled, try with *just* Windows Ink, you are throwing too many parameters at the situation, the exact same parameters when you originally had the issue, nothing is different for us to pinpoint an issue to you.

Debug this yourself by testing after you do something, you will be able to identify what exactly causes it.

fidoboy, 2025-04-22:
Ok. Thanks again.

I've disabled both filters and now keys are working as expected. So there is some issue with the filters. They aren't working as expected. I'll disable then definitively. It could be useful to inform users about issues or incompatibilities when using filters however.

jamesbt365, 2025-04-22:
All plugins are 3rd party supported and we don't provide official support for them, it isn't really feasible for us to know everything about plugins and therefore we don't provide full support for them.

You should figure out which filter it is and report it to the respective GitHub repository.