OpenTabletDriver/OpenTabletDriver #3875: Add support for VEIKK VK1200

twork, 2025-04-21:
### Manufacturer

VEIKK

### Product Name

VK1200

### Device Width

256.32mm

### Device Height

144.18mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

[veikk-1200-diag.json](https://github.com/user-attachments/files/19841062/veikk-1200-diag.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-04-21:
Tablet is not visible on the diagnostic. Send a diagnostic with the tablet plugged in and I'll reopen.

LifeAO, 2025-04-26:
Found this issue as a duplicate for my case. I can provide diags whilst in CachyOS.

**Product page**:
[VEIKK VK1200 - VEIKK](https://veikk.com/route/product/product?product_id=371)

**Diagnostic files**:
[No other drivers - VEIKK1200_otdDiag_noDrivers.json](https://github.com/user-attachments/files/19919501/VEIKK1200_otdDiag_noDrivers.json)
[With VEIKK Linux drivers - VEIKK1200_otdDiag_withDrivers.json](https://github.com/user-attachments/files/19919500/VEIKK1200_otdDiag_withDrivers.json)

**Driver sources**:
Official VEIKK (Debian/RHEL): https://veikk.com/route/v2/support/download
AUR (Arch): [veikk-tablet-bin](https://aur.archlinux.org/packages/veikk-tablet-bin)
> Note: The AUR link directs to a comment where the pkgbuild is to be edited to reflect the recent or latest driver version. The original pkgbuild is outdated.

**Relevant merge requests**:
#3139 - Add support for VEIKK VK1200 - [jamesbt365](https://github.com/jamesbt365)
#3140 - [Backport] Add support for VEIKK VK1200 - [jamesbt365](https://github.com/jamesbt365)

**Other notes**:
The VEIKK VK1200 V2 from #3586 seems to match diagnostics with the VEIKK VK1200. Same serial and friendly name. Unsure of possible conflict.

LifeAO, 2025-04-29:
I just realized that the member that closed this issue is the same whom made the relevant merge requests. 😅
Guess the question is now more of whether another attempt will be entertained now that time has passed since the MR or that the driver dev may be still tricky for the device.