OpenTabletDriver/OpenTabletDriver #3880: Driver doesn't handle windows with admin elevated privileges.

jorgevasquezp, 2025-04-25:
### Steps to Reproduce

1.Switch to any admin-elevated privileges window while using the driver, like task manager or an application that you ran as administrator.
2.Switch back to an application with normal privileges.

### Expected Behavior

Cursor should keep functioning normally even in task-manager or other elevated privileges windows.

### Observed Behavior

When the admin-elevated privileges window is focused the cursor stops responding.
When switching back to a regular window, it continues working as usual.

### Tablet Device

Wacom PTH-660

### Diagnostic Information

[otd_diagnostics.json](https://github.com/user-attachments/files/19915061/otd_diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Kuuuube, 2025-04-25:
This is intended by windows and is not possible to fix.

You can use the windows ink or vmultimode output modes which output kernel level if you wish. Follow the instructions very carefully for setting these up.
https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk
https://github.com/X9VoiD/VoiDPlugins/wiki/VMultiMode

closed 2025-04-25 by Kuuuube

jorgevasquezp, 2025-04-25:
Thanks @Kuuuube for the information, I understand how it is a limitation, but I appreciate the options you present. I'll try them out.