OpenTabletDriver/OpenTabletDriver #3889: Multikey Tablet Button Bindings with Shift Do Not Work on MacOS

JordanSavoie, 2025-05-06:
### Steps to Reproduce

1. Assign a multi-key binding in Auxiliary Settings to a sequence that involves shift (Such as Application+Shit+P, in various drawing apps).
2. Apply the setting and test in app.

### Expected Behavior

It should execute the binding consistently.

### Observed Behavior

Most often the binding doesn't work, but it occasionally does for unclear reasons. Testing in a browser-based keyboard tester shows the corresponding inputs being received, most of the time at least.

This seems similar to Issue #3663, but rather than being an issue with the command key, it's with the shift key.

### Tablet Device

Wacom CTL-4100

### Diagnostic Information

[Untitled.json](https://github.com/user-attachments/files/20070791/Untitled.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-08-27:
@AkiSakurai could you have any idea what this could be?

AkiSakurai, 2025-08-27:
I think it needs some delay before sending each virtual key to the system. If I add a 10ms delay between each key press, the shortcut works more reliably.

gonX, 2025-09-22:
Solved in #4014

closed 2025-09-22 by gonX