OpenTabletDriver/OpenTabletDriver #3892: Add support for Tomate MTM-1106

hennique, 2025-05-12:
### Manufacturer

Tomate

### Product Name

MTM-1106

### Device Width

360

### Device Height

240

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

[This](https://www.tomate.tv/produto/mesa-digitalizadora-614) is the official website for the tablet, I can provide translation if needed.

### Diagnostic Information

[diagnostic.json](https://github.com/user-attachments/files/20153271/diagnostic.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-05-12:
Same boat as #3863 and there is no interest to do such a thing right now as it is extremely tricky to handle this subtree of tablets without causing conflicts and making detection worse on other tablets. I provided a slightly better explanation in the linked issue, but its just too tricky and tedious to be worth our time right now.

closed 2025-05-12 by jamesbt365

gonX, 2025-12-05:
Reopening as we're now interested in trying to fix these, or at least want to track that they're not supported.