OpenTabletDriver/OpenTabletDriver #3893: Support Request: Add stylus support for Huion Kamvas Pro 24 (GT-240)

lvsas, 2025-05-15:
### What do you need help with?

### Device
- Tablet: Huion Kamvas Pro 24 (GT-240)
- Vendor ID: 0x256C
- Product ID: 0x006E
- Parser detected: GianoReportParser

### Issue
The tablet is correctly detected by OpenTabletDriver, but **the stylus is not recognized** at all:
- No pressure is received
- No movement or pen input
- No stylus ID shows up in `detect` logs

Tested with:
- OTD version: 0.6.5.1
- Windows 11 x64
- Clean system (all Huion/Wacom drivers removed via TabletDriverCleanup)

### Additional info
- The stylus and pressure **work perfectly** when using the **official Huion drivers**
- This suggests a missing stylus binding or unsupported input report from this specific Kamvas Pro 24 hardware revision
- I'm happy to help test any custom patch or debug build!

Thank you for your amazing work 💙

### What have you tried?

- Verified that the tablet is properly detected by OpenTabletDriver (device shows up in logs).
- Fully removed all previous Huion and Wacom drivers using TabletDriverCleanup.
- Confirmed stylus pressure works with official Huion drivers (latest version).
- Tried switching USB ports (2.0 and 3.0), no change.

### Diagnostic Information

[odr-report-anonymized-cleaned.json](https://github.com/user-attachments/files/20229438/odr-report-anonymized-cleaned.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

lvsas, 2025-05-15:
Hi, Sorry about the overly cleaned diagnostic file earlier, I went a bit overboard when trying to anonymize things and I understand how that made troubleshooting harder.

That said, your advice was really helpful. I did a full reset to defaults as you suggested, and tested the tablet directly on the desktop. Movement works fine there, but pressure does not.

Then I opened the tablet debugger, and in there, both movement and pressure work perfectly. So it seems the hardware is fine and OTD is receiving the correct data just not passing the pressure through to other applications for some reason.

I assume this might be related to settings like "Windows Ink" or "Wintab", but I’m not sure which one is appropriate for my case.

Let me know if you need a less sanitized version of the JSON or have further advice. 

Thanks for your time!

[image]

jamesbt365, 2025-05-15:
https://opentabletdriver.net/Wiki/FAQ/General#pressure

closed 2025-05-15 by jamesbt365

lvsas, 2025-05-15:
Just wanted to sincerely thank you your guidance helped me solve the issue entirely, and I’m very grateful for your time and clear advice.

Also, it’s reassuring to know that someone as reliable as a certain studious quintuplet is out here helping lost users like me. 😄

Take care, and thanks again!