OpenTabletDriver/OpenTabletDriver #3894: Support for pen twist/rotation data (plumbing, not hardware)

wareya, 2025-05-15:
### Description

hi i have this thing where i can track the twist of my pen using opencv. i want to be able to plumb it into an OTD plugin but OTD seems to not support tablet pen rotation/twist support at all right now (not surprising considering the only hardware that supports it is obscure wacom pens that they literally don't make at all anymore).

this would be cool because there are literally zero desktop tablet setups on currently-manufactured hardware that support twist right now but i could get this jank workaround working if the plumbing was there

This isn't tilt. Twist is the third degree of freedom of rotation after tilt's two degrees of freedom (tiltx/tilty or azimuth/altitude).

https://github.com/user-attachments/assets/0f09f65c-4a1d-4838-8ec8-0604fea7711e

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

wareya, 2025-05-16:
Here I have a proof of concept set up that has a tablet tester connect to my opencv thing over TCP sockets. It's super gross and having it in an OTD plugin would be way better (especially because the application itself wouldn't need to support this bizarre bespoke thing).

https://github.com/user-attachments/assets/3483a90a-a598-4627-9f52-5b62da66245d

Mrcubix, 2025-05-16:
Is the "twist" something the tablet report or something you're just calculating ?

If that's the case, you could make an interface that includes a twist property (`ITwistReport` or call it some other name).
Then if this is an actual tablet, you can build a parser, as well as new reports which may implement this interface.

If done within OTD, you should be able to then just use that data in plugins (by checking for the `ITwitchReport` interface you created).

wareya, 2025-05-16:
In my case, "twist" is something I'm calculating based on the camera feed, but there are tablets that report it. Tablets that fully support the Wacom Art Pen for example. But it's no longer manufactured, and is being scalped on eBay etc. So in practice almost nobody will be able to get this data straight getting the tablet and I'd have no way to test it without generating it from the camera feed.

The windows WM_POINTER Pen API (the one people call "Windows Ink") has a report field for twist, called rotation.

Kuuuube, 2025-05-16:
I have both a Wacom KP-701E and a ZP-600, the two wacom pens that support rotation. I've considered implementing this years ago but never got around to anything complete.

The current parsers actually detect the rotation reports for many tablets (#2737) in order to filter out those reports since they report rotation on the same bits as pressure and cause wonky behavior.