OpenTabletDriver/OpenTabletDriver #3898: Add support for LetSketch WP9620C

AziStacks, 2025-05-20:
### Manufacturer

LetSketch

### Product Name

WP9620C

### Device Width

297mm

### Device Height

190mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

[Diagnostic 5-19-2025.json](https://github.com/user-attachments/files/20322245/Diagnostic.5-19-2025.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

n1ckisthereu, 2025-06-16:
I don't have the same model, but unfortunately I don't have support in opentabletdriver for it.

n1ckisthereu, 2025-06-16:
I have the same model, but unfortunately there is no support in opentabletdriver for it.