OpenTabletDriver/OpenTabletDriver #3905: Add support for ZINNIA MOMENTUM MT1060 TABLET

MCSprime, 2025-05-23:
### Manufacturer

zinnia

### Product Name

momentum mt1060

### Device Width

360

### Device Height

240

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

[DIAGNOSTIC.json](https://github.com/user-attachments/files/20403030/DIAGNOSTIC.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-05-23:
The situation is the same as I told you on Discord, these are too finicky to support as they all conflict with each other and requires additional changes to the way we detect tablets, and we need to figure out a proper way to parse these.

Having lots of requests open for the same effective line of tablets when we have no proper tooling or community to actively support them is a challenge.

for additional information, read #3863 but at this time I'm closing this request.

gonX, 2025-12-05:
Reopening as we're now interested in trying to fix these, or at least want to track that they're not supported.