OpenTabletDriver/OpenTabletDriver #3920: Error popped up after relaunching OTD

PCFucker, 2025-06-14:
### Steps to Reproduce

1. Open Terminal
2. Type "otd-daemon"
3. launch OpenTabletDriver

### Expected Behavior

I expected the program to run so I can map my pen and use my notes app normally

### Observed Behavior

it popped up this message

System.NullReferenceException: Object reference not set to an instance of an object.
   at Eto.GtkSharp.Forms.FormHandler.Show()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at GLib.Idle.IdleProxy.Handler()

### Tablet Device

UGEE Tablet S640U

### Diagnostic Information

[
{
  "Time": "2025-06-14T22:26:02.0852007+03:00",
  "Group": "Plugin",
  "Message": "Loading plugin 'Monitor Toggle'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-14T22:26:02.1190222+03:00",
  "Group": "Detect",
  "Message": "Searching for tablets...",
  "StackTrace": null,
  "Level": 1,
  "Notification": false
},
{
  "Time": "2025-06-14T22:26:02.1375872+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Acepen AP1060'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-14T22:26:02.1396229+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Acepen AP906'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-14T22:26:02.1397072+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Adesso Cybertablet K8'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-14T22:26:02.1401441+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Artisul A1201'",
[... 2355 lines cut ...]
},
{
  "Time": "2025-06-14T22:26:02.4749323+03:00",
  "Group": "UGEE S640",
  "Message": "Tip Binding: [Mouse Button Binding: Left]@1%",
  "StackTrace": null,
  "Level": 1,
  "Notification": false
},
{
  "Time": "2025-06-14T22:26:02.491996+03:00",
  "Group": "Settings",
  "Message": "Driver is enabled.",
  "StackTrace": null,
  "Level": 1,
  "Notification": false
},

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

PCFucker, 2025-08-27:
> We're going to need some more information to work on - which distro is
> this?

On fedora 42 I believe

gonX, 2025-08-27:
The exception is the same as in #3851, also using Fedora. Will track the issue here.

PCFucker, 2025-08-27:
Thank you for letting me know!

closed 2026-11-06 by PCFucker