OpenTabletDriver/OpenTabletDriver #3923: Add Support for macOS 26

advokiller, 2025-06-19:
### What do you need help with?

The first issue is that interface is being displayed incorrectly, as in layers go onto each other.
I also couldn't change any values given.

[image]

### What have you tried?

I tried reinstalling the application in hope that interface displaying will be correct. Before reinstalling my tablet input at least has been detected.

### Diagnostic Information

[Untitled.json](https://github.com/user-attachments/files/20813984/Untitled.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Mrcubix, 2025-06-19:
This is probably an issue with Eto more than OTD.

Could you try fullscreening the window and screenshot how it's layed out ?

gonX, 2026-10-04:
Is this still the case nowadays? #4943 (which will first appear in the next release: v0.6.8) updated the Eto dependencies significantly. Please test a [development build](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=branch%3A0.6.x+workflow%3A%22.NET%20CI%22).