OpenTabletDriver/OpenTabletDriver #3929: Support Request: Xencelabs Pen Tablet Medium Pen Tip Not Detected

leetnope, 2025-06-21:
### What do you need help with?

I am using Linux Mint 22.1 on the Framework 13 AMD Ryzen 7040. Artist mode in tip binding only shows three buttons for the thick pen. It does not show the actual pen tip or settings for the second thinner pen. Thanks for your consideration.

### What have you tried?

I tried reconnecting the tablet via cable and restarting the computer. According to 'https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1535' the tip should be detected.

### Diagnostic Information

SyntaxError: JSON.parse: unexpected end of data at line 2602 column 1 of the JSON data

[
{
  "Time": "2025-06-21T01:48:42.1575376-04:00",
  "Group": "Detect",
  "Message": "Searching for tablets...",
  "StackTrace": null,
  "Level": 1,
  "Notification": false
},
{
Deleted
},
{
  "Time": "2025-06-21T01:48:42.1810271-04:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Huion RTM-500'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-21T01:48:42.1810413-04:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Huion RTP-700'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-21T01:48:42.1810651-04:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Huion WH1409 V2 (Variant 2)'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-21T01:48:42.1810746-04:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Huion WH1409 V2'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-21T01:48:42.1810924-04:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Huion WH1409'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-21T01:48:42.1811031-04:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'KENTING K5540'",
  "StackTrace": null,
  "Level": 0,
[... 1785 lines cut ...]
},
{
  "Time": "2025-06-21T02:12:47.5057105-04:00",
  "Group": "DeviceUnauthorizedAccessException",
  "Message": "Not permitted to open HID class device at /dev/hidraw3.",
  "StackTrace": "   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)\n   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)\n   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()\n   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxOutputReportLength()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_OutputReportLength>b__9_0(HidDevice d) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/dist/debian/opentabletdriver-0.6.5.1/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 26\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/dist/debian/opentabletdriver-0.6.5.1/OpenTabletDriver/Extensions.cs:line 12",
  "Level": 0,
  "Notification": false
},
{
  "Time": "2025-06-21T02:12:47.5060631-04:00",
  "Group": "DeviceUnauthorizedAccessException",
  "Message": "Not permitted to open HID class device at /dev/hidraw3.",
  "StackTrace": "   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)\n   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)\n   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()\n   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxFeatureReportLength()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_FeatureReportLength>b__11_0(HidDevice d) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/dist/debian/opentabletdriver-0.6.5.1/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 27\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/dist/debian/opentabletdriver-0.6.5.1/OpenTabletDriver/Extensions.cs:line 12",
  "Level": 0,
  "Notification": false
},

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-06-21:
Remove the tip binding

leetnope, 2025-06-27:
Sorry for the late reply. Clearing the tip binding does not make the pen tip an option under the eraser settings. The only options are the three pen buttons. If it helps to know, the pen tip is detected by the proprietary driver and I am dual booting with windows.

jamesbt365, 2025-06-27:
Its not gonna make anything appear in the settings but no binding on the tip/eraser means that it should work (no binding = tip/eraser) while you are in artist mode.

leetnope, 2025-06-28:
The cursor does not move when I use the pen tip. This is when I am connecting the tablet to my laptop through a cable.

leetnope, 2025-06-28:
Thank you for the quick response. The cursor does not move when I use the pen tip. This is when I am connecting the tablet to my laptop through a cable.

Also for those who cannot export diagnostics on Linux Mint 22.1, booting into recovery mode (restart -> advanced options for the active linux kernel in the grub menu) and selecting repair damaged packages and updating grub fixed that issue for me. 

Here is my diagnostics file: 
'''json
{
  "App Version": "OpenTabletDriver v0.6.5.1",
  "Build Date": "02/02/2025",
  "Operating System": {
    "Name": "Linux Mint",
    "Version": "22.1 (Xia)",
    "Attributes": {
      "ID": "linuxmint",
      "ID_LIKE": "ubuntu debian",
      "PRETTY_NAME": "Linux Mint 22.1",
      "VERSION_ID": "22.1",
      "VERSION_CODENAME": "xia",
      "KERNEL_VERSION": "6.8.0-62-generic"
    }
  },
  "Environment Variables": {
    "USER": "srouples",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/home/srouples",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/srouples/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.3/0000:c3:00.4/usb7/7-1/7-1:1.2/0003:28BD:5201.0007/hidraw/hidraw6",
      "Manufacturer": "HANVON UGEE",
      "ProductName": "Xencelabs Pen Tablet Medium",
      "SerialNumber": "000000000000",
      "FriendlyName": "Xencelabs Pen Tablet Medium",
      "VendorID": 10429,
      "ProductID": 20993,
      "InputReportLength": 10,
      "OutputReportLength": 33,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORTS": "02:FF0A:0001, 03:FF0A:0001"
      }
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.3/0000:c3:00.4/usb7/7-1/7-1:1.1/0003:28BD:5201.0006/hidraw/hidraw5",
      "Manufacturer": "HANVON UGEE",
      "ProductName": "Xencelabs Pen Tablet Medium",
      "SerialNumber": "000000000000",
      "FriendlyName": "Xencelabs Pen Tablet Medium",
      "VendorID": 10429,
      "ProductID": 20993,
      "InputReportLength": 10,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "1",
        "HID_REPORTS": "07:000D:0002"
      }
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.3/0000:c3:00.3/usb5/5-1/5-1:1.1/0003:32AC:0002.0004/hidraw/hidraw3",
      "Manufacturer": "Framework",
      "ProductName": "HDMI Expansion Card",
[... 2490 lines cut ...]
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2025-06-27T22:06:58.7887368-04:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2025-06-27T22:06:59.5650714-04:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
'''

gonX, 2025-09-02:
Lack of pen tip binding is intended, but the confusion will be reduced with #4010. Closing as wontfix/intended behavior and partially fixed by an upcoming PR.

closed 2025-09-02 by gonX