OpenTabletDriver/OpenTabletDriver #3937: Add support for xoppox d161

SrBolsa, 2025-06-27:
### Manufacturer

xoppox

### Product Name

d161

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

15.6 inch IPS screen
sorry if I don't have enough information about the size

https://www.xoppox.net/Product/showproduct.php?id=28

https://www.xoppox.net/download/download.php?class2=81

### Diagnostic Information

[Diagnostic.5-19-2025.json](https://github.com/user-attachments/files/20936519/Diagnostic.5-19-2025.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

SrBolsa, 2025-08-31:
anyone? please?

gonX, 2025-09-01:
It seems that you have `TabletDriver` installed. Please make sure that it's uninstalled, and then upload another diagnostics.

The first line of the Log part of the diagnostics should tell you whether you have another driver installed.

SrBolsa, 2025-09-01:
[d161.json](https://github.com/user-attachments/files/22069741/d161.json)

like this?, this is an another diagnostic btw