OpenTabletDriver/OpenTabletDriver #3939: Huion H420 Pen Location Interfering with Express Keys

Marckus5, 2025-07-01:
### Steps to Reproduce

## Version
v0.6.5.1
## System Information
System:
  Kernel: 6.12.30+bpo-amd64 arch: x86_64 bits: 64 compiler: N/A Desktop: Cinnamon v: 6.4.8 tk: GTK
    v: 3.24.38 wm: muffin vt: 7 dm: LightDM v: 1.26.0 Distro: LMDE 6 Faye base: Debian 12.1 bookworm
Machine:
  Type: Laptop System: Acer product: Nitro AN515-42 v: V1.18 serial: <superuser required>
  Mobo: RR model: Freed_RRS v: V1.18 serial: <superuser required> UEFI: Insyde v: 1.18
    date: 06/18/2020
Battery:
  ID-1: BAT1 charge: 23.3 Wh (74.2%) condition: 31.4/48.9 Wh (64.3%) volts: 16.4 min: 15.2
    model: LG 0x41,0x43,0x31,0x34,0x42,0x38,0x4B type: Li-ion serial: <filter> status: charging
CPU:
  Info: quad core model: AMD Ryzen 5 2500U with Radeon Vega Mobile Gfx bits: 64 type: MT MCP
    smt: enabled arch: Zen rev: 0 cache: L1: 384 KiB L2: 2 MiB L3: 4 MiB
  Speed (MHz): avg: 1516 high: 1600 min/max: 1600/2000 boost: enabled cores: 1: 1370 2: 1600
    3: 1371 4: 1496 5: 1559 6: 1583 7: 1552 8: 1599 bogomips: 31938
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm

  Display: x11 server: X.Org v: 1.21.1.7 with: Xwayland v: 22.1.9 driver: X: loaded: amdgpu
    dri: radeonsi gpu: amdgpu display-ID: :0 screens: 1
  Client: Cinnamon v: 6.4.8 inxi: 3.3.26

## Steps
1. Connect Huion H420 Tablet
2. Open Tablet Debugger
3. See Image Below:
[image]
   a. Pen Condition (Not Near, Hovering, Touching Tablet, etc.)
   b. Pen Location When near tablet
   c. Pen Pressure
   d. Express Keys when pen not near tablet

### Expected Behavior

Pen location is reset to zero when put away, making sure all express keys register as not pressed.

### Observed Behavior

Pen location does not reset, leading to situations where the express keys are sometimes registered as pressed. Physically pressing the keys does reset the values.

### Tablet Device

Huion H420

### Diagnostic Information

[diag_opentabletdriver.json](https://github.com/user-attachments/files/20989597/diag_opentabletdriver.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

jamesbt365, 2025-07-02:
Unless huion silently changed something, i'm guessing this is a firmware bug or other quirk of the tablet, could also just be a bunch of interference around, have you tried with the stock driver yet?

Marckus5, 2025-07-02:
> Unless huion silently changed something, i'm guessing this is a firmware bug or other quirk of the tablet, could also just be a bunch of interference around, have you tried with the stock driver yet?

I tried it with a Windows 11 computer (this model doesn't have official Linux drivers). The original driver does not have this issue.

I tested it with MS Whiteboard. Using OpenTableDriver with the three express keys (AuxButtons inside OpenTabletDriver) set to ctrl+z (undo), I hovered the pen above the tablet and kept lifting it away at random spots. It would register the undo key whenever I lifted the pen at certain y coordinates (consistent with the raw tablet data at the debugger). I tried this with the original driver, and it does not exhibit this behavior.

gonX, 2025-09-02:
Please provide a data recording - it's done via the Tablet Debugger, and the file is named `tablet-data.txt` and exists in the application data directory.