OpenTabletDriver/OpenTabletDriver #3941: Add support for

JesusKarma, 2025-07-02:
### Manufacturer

Wacom

### Product Name

CTL-472

### Device Width

58

### Device Height

45

### Additional Information

I can't install the plugin, I click on install and the program just freezes and gives this https://ibb.co/LXrXf8sG

### Diagnostic Information

[ffff.json](https://github.com/user-attachments/files/21024951/ffff.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

jamesbt365, 2025-07-02:
Use the discord server, you are misusing our github issues templates, but this is likely a internet issue

https://discord.gg/9bcMaPkVAR

gonX, 2025-08-27:
Please use the support request template a make a new issue if you're still having an issue.

closed 2025-08-27 by gonX