OpenTabletDriver/OpenTabletDriver #3944: Support Request: My huion 420 is being recognized but isnt actually working

Nikorasu038, 2025-07-04:
### What do you need help with?

so yesterday when i first tested it it was working fine, i could set everything and even use it on osu but today for some reason it just doesnt work, my pen in the debug is registring but my mouse doesnt move

I use arch linux

### What have you tried?

tried re-installing otd and tried some commands from the AUR 

### Diagnostic Information

[2025-07-04 12_52_55Z.log.json](https://github.com/user-attachments/files/21065568/2025-07-04.12_52_55Z.log.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Nikorasu038, 2025-07-04:
> > I use arch linux
> 
> Is this all the system information you are going to provide? you haven't provided a complete diagnostic either so seeing stuff about the usb devices attached to your computer is impossible too.
> 
> Please explain other information about your system including information about what Desktop environment/Window manager you are using and other useful diagnostic information, or we will be unable to help you.
> 
> To get a complete diagnostic file, go to help > export diagnostics, save that file and send it here.

1. im not sure what info you need, but if its about cable connection i don't believe thats the case bc both arch and otd recognizes it
 2. i use hyprland
3. idk how but when i try to get the files it just opens dolphin
also im a newbie on both arch/hyprland so if you could help me find more info id be greatful

gonX, 2026-10-04:
Closing stale support request

closed 2026-10-04 by gonX