OpenTabletDriver/OpenTabletDriver #3950: XP-Pen Artist 12 Pro X/Y size may be wrong or device has multiple revisions.

darrencocco, 2025-07-11:
### Steps to Reproduce

1. Have XP-Pen Artist 12 Pro
2. Use default configuration of OTD
3. Hold pen at top left hand corner
4. Move pen diagonally to bottom right corner

### Expected Behavior

When moving pen cursor should stay roughly aligned with pen position

### Observed Behavior

In top left hand corner pointer is aligned as we move further away the pointer starts to race ahead until you reach the middle of the tablet with the pen at which the pointer has moved all the way to the bottom right corner.

### Tablet Device

XP-Pen Artist 12 Pro

### Diagnostic Information

Looking the data in the OTD Tablet Debugger the X/Y co-ordinates go all the way to 52593, 29614. This is approximately double what was in the configuration added by this pull (https://github.com/OpenTabletDriver/OpenTabletDriver/pull/725)

I've attached the export debug with some sensitive data removed.

[xp-pen_artist_12_pro.json](https://github.com/user-attachments/files/21175842/xp-pen_artist_12_pro.json)

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-08-27:
For anyone with an affected tablet, please upload a full String Dump (via Tablets -> Device String Reader). If that doesn't work, we likely need string 1, 2, 3, 4, 5 and 201.

Mrcubix, 2026-10-30:
Index 100 could probably be used for identification.

1. Go to `File > Tablet debugger`,
2. Select your device `UGTABLET 11.6 inch PenDisplay`,
3. For the index, use `100`,
4. Click on `Send Request`,
5. Provide the result here.