OpenTabletDriver/OpenTabletDriver #3951: Add support for Huion Kamvas 16 (Gen 3) (almost already present?)

IHAVENOARMS, 2025-07-13:
### Manufacturer

Huion

### Product Name

Kamvas 16 (Gen 3)

### Additional Information

I saw that there was support for Huion Kamvas 13 (Gen 3) on the website, the Huion Kamvas 16 (Gen 3) is the exact same, the screen is just larger (16 inches, hence the name). You can find the tablet below under the ProductName Huion Tablet_GS1563.

### Diagnostic Information

```json
{
  "AppVersion": "OpenTabletDriver v0.7.0.0-pre",
  "BuildDate": "",
  "OperatingSystem": {
    "Name": "Arch Linux",
    "Version": "Unknown",
    "Attributes": {
      "PRETTY_NAME": "Arch Linux",
      "ID": "arch",
      "BUILD_ID": "rolling",
      "KERNEL_VERSION": "6.15.5-arch1-1"
    }
  },
  "EnvironmentVariables": {
    "USER": "jimmy",
    "DISPLAY": ":1",
    "WAYLAND_DISPLAY": "wayland-1",
    "PWD": "/home/jimmy",
    "PATH": "/home/jimmy/.pyenv/shims:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "Devices": [
    {
      "VendorID": 18003,
      "ProductID": 4,
      "InputReportLength": 9,
      "OutputReportLength": 2,
      "Manufacturer": "foostan",
      "ProductName": "Corne v4",
      "FriendlyName": "Corne v4",
      "SerialNumber": "vial:f64c2b3c",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.2/1-1.2.3/1-1.2.3:1.0/0003:4653:0004.0009/hidraw/hidraw8",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS": "00:0001:0006"
      }
    },
    {
      "VendorID": 9580,
      "ProductID": 8201,
      "InputReportLength": 14,
      "FeatureReportLength": 8,
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1563",
      "FriendlyName": "Huion Tablet_GS1563",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:256C:2009.0015/hidraw/hidraw16",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS": "08:FF00:0001"
      }
    },
    {
      "VendorID": 21972,
[... 220 lines cut ...]
    {
      "Time": "2025-07-13T20:22:18.0630033+03:00",
      "Group": "IPC",
      "Message": "Connected to a client."
    },
    {
      "Time": "2025-07-13T20:22:25.3678238+03:00",
      "Group": "Detect",
      "Message": "Searching for tablets...",
      "Level": 1
    },
    {
      "Time": "2025-07-13T20:22:25.3740791+03:00",
      "Group": "Detect",
      "Message": "No tablets were detected.",
      "Level": 1
    }
  ]
}
```

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

IHAVENOARMS, 2025-08-11:
I’m working on adding it myself I’m 90% of the way there but still no wheels support, will make a PR soon, Although I think devs stopped merging PRs to master so you’ll have to use the other branches or you can use my fork when I push the changes there, will update you when that happens

gonX, 2025-09-19:
> I think devs stopped merging PRs to master

If you're making a PR, please just target `0.6.x` or `avalonia` and we will ensure it will make it into the next release. We don't merge PR's to master since it increases the amount of work we have to do once the `avalonia` branch takes over as the new `master` at a later point.