OpenTabletDriver/OpenTabletDriver #3987: Add support for XP-Pen Artist 15.6 Pro (V2)

Apollo11MyGuy, 2025-08-17:
### Manufacturer

XP-Pen

### Product Name

XP-Pen Artist 15.6 Pro (V2)

### Device Width

442.91mm  - Active display: 340.99mm

### Device Height

279.91mm - Active display: 191.81 mm

### Pressure Levels

16383

### Lines per Inch (LPI)

5080

### Diagnostic Information

[help.json](https://github.com/user-attachments/files/21823810/help.json)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

Apollo11MyGuy, 2025-08-17:
it does not register as a tablet at all

AndrewLinden, 2025-08-27:
I got my XP-Pen Artist 15.6 Pro (V2) delivered today. And apart from the auxiliary buttons and the wheel I got it to work. See my config file
 

[Artist 15.6 Pro (Gen2).json](https://github.com/user-attachments/files/22013485/Artist.15.6.Pro.Gen2.json)