OpenTabletDriver/OpenTabletDriver #3996: Support Request: Discord invite is invalid

KryoKillz, 2025-08-21:
### What do you need help with?

I need help since opentabletdriver isnt detecting my tablet as was going to go to the discord for help but the invite is expired or something. Can you send a new invite to the server or something please.

### What have you tried?

refreshing the page and searching up reasons on why the invite was invalid and it said that its possible that the invite is expired.

### Diagnostic Information

[diagnostics.json](https://github.com/user-attachments/files/21925940/diagnostics.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

Kuuuube, 2025-08-21:
The invite on the OTD site (https://opentabletdriver.net/) is still valid. Where did you find the invalid one?

Here is the current invite linked to on the OTD site https://discord.com/invite/9bcMaPkVAR.

closed 2025-08-21 by Kuuuube

KryoKillz, 2025-08-21:
[image]
Maybe it's something on my end but it wont let me join