OpenTabletDriver/OpenTabletDriver #4053: [06x] Plugin Manager suggests drag and drop to install plugins, even when not supported

gonX, 2025-09-07:
### Steps to Reproduce

1. Use Linux
2. Open Plugin Manager
3. Observe that the bottom of the window says "Drag and drop plugins here to install."
4. Try to drag a plugin zip in
5. Window converts to "Drag and drop is not supported on this platform"

### Expected Behavior

The plugin manager should not suggest the user to drag and drop files if it isn't supported

### Observed Behavior

The bottom of the window says "Drag and drop plugins here to install" despite not being supported on my platform (Linux)

### Tablet Device

n/a

### Diagnostic Information

n/a but version is latest 0.6 on 7th sept 2025: 706bd9e3e793455297a6a53e0325507109fc9824

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-09-20:
Moving milestone of this issue to 0.6.7 given the comment by @Mrcubix suggesting it's fixed by updating Eto, which was planned for 0.6.7.

Danielr19, 2026-09-01:
Hi I'd like to contribute by solving this issue.