OpenTabletDriver/OpenTabletDriver #4073: Crash when clicking on "..." for Tip Binding after uninstalling VMultiMode

godfatherjohn, 2025-09-13:
### Steps to Reproduce

Usually I use Windows Ink Absolut but for some reason I decided to try VMultiMode ... installed it, then uninstalled it. Now the following error message pops up whenever I try to change my tip binding. Currently, OTD is not working, the tip binding says ":Button: Left" which is left over from VMultiMode. (I will remove OTD and start anew.)  The software told me to inform the programmers about his issue. 

This very likely is not tablet dependent. 

System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.Controls.PluginSettingStoreEditor`1.GetControlsForType(PluginSettingStore store, Type type)+MoveNext() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\PluginSettingStoreEditor.cs:line 84
   at System.Linq.Enumerable.ConcatIterator`1.MoveNext()
   at OpenTabletDriver.UX.Controls.PluginSettingStoreEditor`1.OnStoreChanged() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\PluginSettingStoreEditor.cs:line 43
   at OpenTabletDriver.UX.Controls.PluginSettingStoreEditor`1.set_Store(PluginSettingStore value) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\PluginSettingStoreEditor.cs:line 29
   at OpenTabletDriver.UX.Windows.Bindings.AdvancedBindingEditorDialog..ctor(PluginSettingStore currentBinding) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Windows\Bindings\AdvancedBindingEditorDialog.cs:line 75
   at OpenTabletDriver.UX.Controls.BindingDisplay.<.ctor>b__0_2(Object sender, EventArgs e) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\BindingDisplay.cs:line 46
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

### Expected Behavior

I should be able to select my tip binding option.

### Observed Behavior

Cannot change the tip binding after uninstalling VMultiMode.

### Tablet Device

Kamvas Pro 2.5K 

### Diagnostic Information

System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.Controls.PluginSettingStoreEditor`1.GetControlsForType(PluginSettingStore store, Type type)+MoveNext() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\PluginSettingStoreEditor.cs:line 84
   at System.Linq.Enumerable.ConcatIterator`1.MoveNext()
   at OpenTabletDriver.UX.Controls.PluginSettingStoreEditor`1.OnStoreChanged() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\PluginSettingStoreEditor.cs:line 43
   at OpenTabletDriver.UX.Controls.PluginSettingStoreEditor`1.set_Store(PluginSettingStore value) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\PluginSettingStoreEditor.cs:line 29
   at OpenTabletDriver.UX.Windows.Bindings.AdvancedBindingEditorDialog..ctor(PluginSettingStore currentBinding) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Windows\Bindings\AdvancedBindingEditorDialog.cs:line 75
   at OpenTabletDriver.UX.Controls.BindingDisplay.<.ctor>b__0_2(Object sender, EventArgs e) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\BindingDisplay.cs:line 46
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-09-20:
I cannot seem to reproduce this on 4312b4a1ccedec2b4c52066838e1c033c22e97a8

This means that it should be solved with the next release.

closed 2025-09-20 by gonX

gonX, 2025-09-21:
Like solved by #3783 by the way