OpenTabletDriver/OpenTabletDriver #4081: Add support for DTK-2420 (Cintiq Pro 24", 2018)

corigne, 2025-09-17:
### Manufacturer

Wacom

### Product Name

DTK-2420

### Device Width

523

### Device Height

294.2

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Working configs for this have been posted to the discord.

### Diagnostic Information

```json
{
  "AppVersion": "OpenTabletDriver v0.7.0.0-pre",
  "BuildDate": "",
  "OperatingSystem": {
    "Name": "Arch Linux",
    "Version": "Unknown",
    "Attributes": {
      "PRETTY_NAME": "Arch Linux",
      "ID": "arch",
      "BUILD_ID": "rolling",
      "KERNEL_VERSION": "6.16.7-arch1-1"
    }
  },
  "EnvironmentVariables": {
    "USER": "nexus",
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-1",
    "PWD": "/home/nexus",
    "PATH": "/run/user/1000/fnm_multishells/1256_1758138534685/bin:/home/nexus/.local/share/fnm:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "Devices": [
    {
      "VendorID": 1133,
      "ProductID": 49295,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Logitech",
      "ProductName": "G403 HERO Gaming Mouse",
      "FriendlyName": "G403 HERO Gaming Mouse",
      "SerialNumber": "178534503936",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:0e:00.3/usb5/5-4/5-4:1.1/0003:046D:C08F.0005/hidraw/hidraw4",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "1",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 1386,
      "ProductID": 817,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "ExpressKey Remote ",
      "FriendlyName": "ExpressKey Remote ",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3:1.0/0003:056A:0331.0003/hidraw/hidraw2",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 6940,
      "ProductID": 3088,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Corsair",
      "ProductName": "Commander PRO",
      "FriendlyName": "Commander PRO",
      "SerialNumber": "0A05009C93270C1F",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-6/1-6:1.0/0003:1B1C:0C10.0007/hidraw/hidraw5",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 17498,
      "ProductID": 8800,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "KBDFans",
      "ProductName": "DZ60",
      "FriendlyName": "DZ60",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-2/1-2.1/1-2.1:1.1/0003:445A:2260.0008/hidraw/hidraw7",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "1",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 3034,
      "ProductID": 16520,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Generic",
      "ProductName": "USB Audio",
      "FriendlyName": "USB Audio",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:0e:00.3/usb5/5-3/5-3.2/5-3.2.4/5-3.2.4:1.2/0003:0BDA:4088.0009/hidraw/hidraw8",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 2821,
      "ProductID": 6387,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "AsusTek Computer Inc.",
      "ProductName": "AURA LED Controller",
      "FriendlyName": "AURA LED Controller",
      "SerialNumber": "9876543210",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6:1.2/0003:0B05:18F3.0002/hidraw/hidraw1",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 1133,
      "ProductID": 49295,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Logitech",
      "ProductName": "G403 HERO Gaming Mouse",
      "FriendlyName": "G403 HERO Gaming Mouse",
      "SerialNumber": "178534503936",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:0e:00.3/usb5/5-4/5-4:1.0/0003:046D:C08F.0004/hidraw/hidraw3",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 1386,
      "ProductID": 892,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Wacom Cintiq Pro 24",
      "FriendlyName": "Wacom Cintiq Pro 24",
      "SerialNumber": "8JH00F1003391",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:0e:00.3/usb5/5-3/5-3.5/5-3.5:1.0/0003:056A:037C.000A/hidraw/hidraw9",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 17498,
      "ProductID": 8800,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "KBDFans",
      "ProductName": "DZ60",
      "FriendlyName": "DZ60",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-2/1-2.1/1-2.1:1.0/0003:445A:2260.0006/hidraw/hidraw6",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 1133,
      "ProductID": 2195,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Logitech StreamCam",
      "FriendlyName": "Logitech StreamCam",
      "SerialNumber": "A8A43645",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb4/4-4/4-4:1.5/0003:046D:0893.0001/hidraw/hidraw0",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "5",
        "HID_REPORTS_NON_RECONSTRUCTABLE": "true"
      }
    }
  ],
  "ConsoleLog": [
    {
      "Time": "2025-09-17T14:48:57.4205758-05:00",
      "Group": "RootHub",
      "Message": "Initialized internal child hubs: HidSharpDeviceRootHub"
    },
    {
      "Time": "2025-09-17T14:48:57.6258829-05:00",
      "Group": "Detect",
      "Message": "Searching for tablets...",
      "Level": 1
    },
    {
      "Time": "2025-09-17T14:48:57.630036-05:00",
      "Group": "Detect",
      "Message": "No tablets were detected.",
      "Level": 1
    },
    {
      "Time": "2025-09-17T14:49:02.2910814-05:00",
      "Group": "IPC",
      "Message": "Connected to a client."
    },
    {
      "Time": "2025-09-17T14:49:13.0801793-05:00",
      "Group": "Detect",
      "Message": "Searching for tablets...",
      "Level": 1
    },
    {
      "Time": "2025-09-17T14:49:13.0874135-05:00",
      "Group": "Detect",
      "Message": "No tablets were detected.",
      "Level": 1
    },
    {
      "Time": "2025-09-17T14:49:26.0861162-05:00",
      "Group": "Detect",
      "Message": "Searching for tablets...",
      "Level": 1
    },
    {
      "Time": "2025-09-17T14:49:26.0863321-05:00",
      "Group": "Detect",
      "Message": "No tablets were detected.",
      "Level": 1
    },
    {
      "Time": "2025-09-17T14:49:28.2497755-05:00",
      "Group": "IPC",
      "Message": "Disconnected from a client."
    },
    {
      "Time": "2025-09-17T14:59:50.710864-05:00",
      "Group": "IPC",
      "Message": "Connected to a client."
    }
  ]
}
```json

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

gonX, 2025-09-21:
Please provide a screenshot of the tablet debugger with the Discord configuration showing the maximum reported position. The `MaxX` and `MaxY` values in the configuration posted on Discord does not seem correct to me.

[Discord Thread](https://canary.discord.com/channels/615607687467761684/1349165694142517309)

corigne, 2025-09-22:
I will do so shortly. Wrt the max values, I think one of the users in the corresponding thread had incorrect values, and another user provided correct values later in the thread. Should have it up today or tomorrow at theatest.