OpenTabletDriver/OpenTabletDriver #4092: Add support for Deco 640

baggee134-prog, 2025-09-20:
### Manufacturer

XP-Pen

### Product Name

Deco 640

### Device Width

160mm

### Device Height

90mm

### Pressure Levels

16384

### Lines per Inch (LPI)

5080 

### Additional Information

It doesn't appear on OpenTabletDriver

### Diagnostic Information

```json
 "App Version": "OpenTabletDriver v0.6.5.1",
  "Build Date": "2/2/2025",
  "Operating System": {
    "Name": "Win32NT",
    "Version": "10.0.26100.0",
    "Attributes": {
      "ServicePack": ""
    }
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\bagge\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\bagge\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\bagge"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_0408&pid_5489&mi_02#6&281d0fdf&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "Quanta ",
      "ProductName": "HP True Vision HD Camera",
      "SerialNumber": "0001",
      "FriendlyName": "HP True Vision HD Camera",
      "VendorID": 1032,
      "ProductID": 21641,
      "InputReportLength": 0,
      "OutputReportLength": 0,
      "FeatureReportLength": 61,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORTS": "20:FF07:0031, 22:FF07:0031, 25:FF07:0031"

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.

WangKe-117, 2025-09-21:
Honestly, I also really hope that OpenTabletDriver will support this tablet soon. It seems compact, affordable, and quite suitable for playing osu!

gonX, 2025-09-22:
Duplicate of #3373

closed 2025-09-22 by gonX