OpenTabletDriver/OpenTabletDriver #4124: Can you adjust

amrmenam, 2025-09-25:
### Description

Is it possible to make the hand erase directly without pressing the eraser and make only the pen that writes and the finger scrolling is there an addition or modification that can be done this
i am working on interactive panel whiteboard

[image]
[image]

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

gonX, 2025-09-25:
We don't have touch parsed at the moment, see #1664 

Windows Ink already has an eraser button feature, if you're on Windows I suggest using that. For Linux and MacOS the please see my concerns in [#2617](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2617#issuecomment-1416533886)

It is unclear to me what you're requesting exactly.

amrmenam, 2025-09-26:
in this video https://youtu.be/EZUoqNChUKE?t=3 he using smartboard activepanel its using smart ink its work only on smartboard . the hand erase directly without pressing the eraser and make only the pen that writes and the finger scrolling

closed 2025-09-30 by gonX