OpenTabletDriver/OpenTabletDriver #4134: Air brush's wheel support

IrisRealtime, 2025-09-27:
### Description

Wacom Air brush (ZP-400) is an Intuos3 and Cintiq 1st Gen compatible pen, made to resemble a real analog brush of the same name. Beside the button up top, ZP-400 has a wheel that spins 90 degrees, where the real brush has a lever, which is supposed to imitate changing the spread of paint. 
With the official drivers, you are not able to change what keypress each direction on the airbrush is performing and the user has to configure its behavior in their professional suite of choice, given it's compatible.
Ideal support implementation for this wheel would be an ability to map any button, click or scroll direction (if possible) to each direction of scrolling.
[Product page](https://estore.wacom.com/en-au/intuos3-cintiq-1st-gen-air-brush.html?srsltid=AfmBOopVXrTAq0H8vH1qDUbMo4ezujarvuoIWz_fkSb6DEGHdcqpXirQ)

### Acknowledgements

- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

gonX, 2025-12-29:
This needs a tablet data recording of these wheel motions. Please open the tablet debugger inside of OpenTabletDriver and start data recording, then bring your pen in and click all possible buttons and move any relevant gadgets, and then upload the recording file here. It will be in the [appdata](https://opentabletdriver.net/Wiki/FAQ/General#appdata) folder

gonX, 2026-01-02:
In case we don't get back to this for a while, the filename from the above download is `tablet-data_1759631539.txt` and may be in the downloads folder of both Kuuube and I.