OpenTabletDriver/OpenTabletDriver #4164: Auxiliary buttons not working

L1teD, 2025-09-30:
### Steps to Reproduce

1. Check in tablet debugger that auxiliary buttons are detected

[image]

2. Try to bind anything to them

[image]

3. Get depressed when you see that they don't work

### Expected Behavior

I expected any of binding types to work (Multi-key, Key, Mouse, etc.)

### Observed Behavior

Nothing happens at all, tablet debugger and tablet are reacting to presses, but actual bindings not working

### Tablet Device

CTH-480

### Diagnostic Information

[diagnostic.json](https://github.com/user-attachments/files/22609525/diagnostic.json)

### Acknowledgements

- [x] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [x] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [x] I have searched the existing issues and this new issue is not a duplicate of any.
- [x] I have written a concise and meaningful title.
- [x] I am on the latest version of OpenTabletDriver.

L1teD, 2025-09-30:
nevermind, it's just decided to work after two weeks

L1teD, 2025-09-30:
actually no, it seems they're working only in windows ink mode, once switched back to default absolute mode, they stopped working
also in windows ink mode, plugins like preset/scroll/etc. bindings not working, only mouse and key bindings work

gonX, 2025-10-01:
The tablet debugger screenshot looks like you're holding down all 4 auxiliary keys. Please confirm that all 4 show `False` when nothing is held, and each individual key should cause a single change in `AuxButtons` state.

gonX, 2025-10-06:
Please upload a tablet data recording of you pressing all 4 keys individually. You can enable it in the tablet debugger, and the resulting file will end up in your application data folder, file name starting with `tablet-data`

L1teD, 2026-02-16:
So after a time i found that the issue is in Temporal Resampler, for some reason when i turn it on, all aux buttons stops working completely, but once i turn it off, all is good